circuitdiagram/components #9: Ability to have super- and sub-script text

aumars, 2018-09-15:
It would be nice to be able to add super- and sub-script text into the text field of components.

SamuelFisher, 2018-09-26:
Hi, you can already do this when creating new components - see here for an example:

https://github.com/circuitdiagram/components/blob/238998c6c17c7328866e7bda8915919fa2604dcd/Components/ICs/microcontroller.xml#L69

Or do you mean allowing this when entering text into a property on an existing component?

aumars, 2018-09-26:
I would like to use super- and sub-script text into a property of an already existing component.

JambonSama, 2019-08-04:
@aumars Some unicode characters are already supported, like this 
[image] if that helps.
That's really, really cool, btw (props to whoever coded that right).

SamuelFisher, 2024-03-20:
You can now make subscript text by selecting the "Text Formatting" shader when exporting a circuit (it won't show subscript in the editor itself though). More information about this is here:

https://www.circuit-diagram.org/news/2024/03/introducing-circuit-shaders

and here's an example:

[image]