circuitdiagram/components #10: Pneumatic line

eplotner, 2018-10-17:
May not be completely in the scope of this but a few pneumatic components would be nice.

Corsaka, 2019-03-31:
These?
[image]

eplotner, 2019-04-01:
Yes.

[image]<https://user-images.githubusercontent.com/23699128/55291485-ddc08800-53d6-11e9-9704-beb86b728106.png>

WARNING: All email sent to or from this address will be received or otherwise recorded by the Bel Fuse Inc. corporate email system and is subject to archival, monitoring or review by, and/or disclosure to, someone other than the recipient.

SamuelFisher, 2019-04-01:
I think this is a tricky one because it seems like pneumatic components are distinct to electronic components, and they really belong to a different type of diagram which is so far not what Circuit Diagram has aimed to provide.

I also don't think it makes sense to be able to use both types of components in the same diagram.

closed 2020-11-08 by SamuelFisher