circuitdiagram/components #54: VSCode extension unable to find executable

Corsaka, 2019-08-04:
Current path:
{     "circuitDiagram.executablePath": "C:\Program Files\Circuit Diagram Custom Content\circuit-diagram-cli.exe" }

It's also been tried with forward slashes, two backward slashes, and two forward slashes.

I don't know what else could be the issue here.

SamuelFisher, 2019-08-04:
Please could you try running the following in PowerShell to check there are no special characters in the path?

```ps
Test-Path "C:\Program Files\Circuit Diagram Custom Content\circuit-diagram-cli.exe"
```

Corsaka, 2019-08-05:
[image]

SamuelFisher, 2019-08-05:
Thanks. Setting that with two backward slashes should do the right thing, but it's odd that didn't work.

```
{ "circuitDiagram.executablePath": "C:\\Program Files\\Circuit Diagram Custom Content\\circuit-diagram-cli.exe" }
```

Are you running Windows 32 or 64-bit?

You can check using the following PowerShell command:

```
> gwmi win32_operatingsystem | select OSArchitecture

OSArchitecture
--------------
64-bit
```

Corsaka, 2019-08-05:
64-bit. I'll try reinstalling the vsix and reconfiguring the settings.

closed 2019-08-18 by SamuelFisher