circuitdiagram/components #55: Text instead of value for all / more components

JambonSama, 2019-08-04:
Some components can display texts, but some can only display a value, or even nothing : 

- the resistor has so many cool fields : 
[image]
I can display the value or the text, and their is only one component but I can choose between all the way of representing a resistor (so, obviously the field 'IEC / US' and the field 'type' are awesome).

- the inductor is a little more tricky : 
[image]
I would ideally like to have similar options of writing a value and a text, and displaying one or the other (or both ? maybe ?) and proposing the different ways of displaying an inductor (either the looping thing or the filled rectangle for example) would be really nice.

- and finally the motor : 
[image]
Yeah, the image speaks for itself : the possibility to add text besides the component would be super awesome.

I'm writing this from a place of "I want to draw theoretical diagrams for papers and reports, and sometimes those need to be commented, but the values don't always matter depending on what you want to show".

Corsaka, 2019-08-05:
There's a label "component". Use that.

JambonSama, 2019-08-05:
Nice, thanks, I should have checked that before bothering you. 
But still, what if I want the value field of my component not to show, then ?
[image]

SamuelFisher, 2019-08-05:
Hi @JambonSama, thanks for highlighting these issues and for the suggestion of more display styles for the inductor component.

For now I have added support for the inductor to display text instead of the inductance value:

[image]

For the motor it's probably best to use a label to add text.

Please let me know if there are any issues.

JambonSama, 2019-08-05:
Thank you very much!

closed 2019-08-05 by JambonSama