circuitdiagram/components #69: Generator

15-ai, 2020-01-11:
It would be very helpful to have a generator circuit symbol to use.

SamuelFisher, 2020-01-12:
Hi, does this include the type you need?

[image]

15-ai, 2020-01-12:
Hi,
Thanks for your response, it does include the type of generator that I need.
Could you please tell me where I can find these to add on to the Circuit Diagram program,
since I’m new to the program and using it for the first time.

[image]<https://user-images.githubusercontent.com/5303088/72216218-4d42d680-3516-11ea-8008-99eed7e312ce.png>

SamuelFisher, 2020-01-12:
This has been added in 4ce4e6674d436fecf0b7832f611b2d70e3ae3878.

It will be available to use in the [Web Editor]() shortly (please allow a bit of time for it to appear) by using the _Add more components_ button:

[image]

To use in Circuit Diagram Desktop, download the CDCOM file from [here](https://www.circuit-diagram.org/components/15c74cca-a629-47f8-8b11-67ca282523ad) and follow [these instructions](https://www.circuit-diagram.org/help/installing-components).

closed 2020-01-12 by SamuelFisher

15-ai, 2020-01-12:
Thank you very much for the quick response as I needed the software for a school project which is due very soon.
And I’m sorry for bothering you with something which was already there but this is my first time using this program.

Kind regards,
15-ai

[image]<https://user-images.githubusercontent.com/5303088/72218579-1e3b5d80-3534-11ea-871f-a06e8b8a466c.png>