circuitdiagram/components #75: [Request] Add crossing wires symbol

mm2595, 2020-02-21:
Please add "connected" and "not connected" symbols.

Corsaka, 2020-02-21:
...not connected = no connection visible.
[image]

connected = connection visible.
[image]

mm2595, 2020-02-27:
Hi,

Thanks for your reply. I am not quite sure how to make connections
invisible. Every time I make two wires crossed, there is a dot. Can you
help me with that?

Corsaka, 2020-02-27:
Connections will always be shown by a dot. If the wires are not connected, there will not be a dot. To overlay two wires without the dot showing, make sure that the end of one wire is not on the other one.

dyllbert, 2020-02-27:
While I agree it does implement this feature, it would be nice for clarity to be able to manually insert the rounded "not connected" symbol something like this:

[image]

SamuelFisher, 2020-02-29:
I don't think this fits as a "component" however I will leave this issue open to keep track of this feature.