circuitdiagram/components #76: Perfecting the IC component

JagodaT, 2020-02-22:
I think two quick fixes to the Integrated Circuit component would be much appreciated! Putting the little identation on the "casing" as to signal "up" and "down" positions; and, also, reassigning the PIN editor to the standard pin numbering in IC's, ie, pin 1 starting on the top left and the numbering going counter-clokwise

portrf, 2021-01-30:
I agree, the present numbering system which starts at 0 and changes sides is very odd.

PaulH463, 2021-11-19:
I agree with those suggestions, also If the IC could be rotated 90 degrees that would be really nice too.

G1OJS, 2023-08-05:
I've just done a lot of this for my personal use with the desktop app. Just started with the common sizes of 8,14,16 and 18 pin DIL & added the notch, started at Pin 1 not Pin 0 and used the conventional anticlockwise numbering. Happy to share the XML but pull requests don't seem to be looked at very often so that really doesn't seem the best route?