circuitdiagram/components #78: Subtext in component creation

Corsaka, 2020-02-27:
Right now I'm trying to make a section of text that is a lot smaller than the rest in order to separate it clearly, using the `size` option. However, this doesn't seem to work, as this:
```xml
<text x="_Start-5" y="_Middle" align="CentreCentre" size="large" value="C" />
<text x="_Start+5" y="_Middle" align="CentreCentre" size="small" value="in" />
```
Turns out as this:
[image]
Is this at all possible, or is there some other way to indicate subtext I'm unaware of?

SamuelFisher, 2020-02-27:
You can create a single text element that itself contains multiple text runs. Each text run can be one of the following: _span_ (normal text), _sub_ (subscript) or _sup_ (superscript).

See here for an example:

https://github.com/circuitdiagram/components/blob/e84e7a6bc457a8fd1139e7aa43227a591c7438a8/ic/microcontroller/microcontroller.xml#L67-L72

[image]

Corsaka, 2020-02-27:
So the `size` modifier is unnecessary? I've tried both "small" and "medium" and nothing seems to change.

SamuelFisher, 2020-02-27:
You don't need to use `size` to make the text smaller when using `sub`/`sup`. The only option for size is `size="large"`. If you don't specify it defaults to normal/medium size.

closed 2020-02-27 by SamuelFisher