circuitdiagram/components #101: Please add generic optocouplers

papilinou, 2020-11-04:
A standard electronic component is missing : optocoupler.
Please add it in 4,6,8 pins.

Corsaka, 2020-11-05:
Can't you just combine an LED and a phototransistor?

papilinou, 2020-11-08:
yes,that's what i do at each time i need one.I'm now studying component design and near to set a pull request.I'll test it today on CD desktop.
The only thing not clear is how to get an GUID when using gitpod.

SamuelFisher, 2020-11-08:
> The only thing not clear is how to get an GUID when using gitpod.

You can create a GUID by running this command in the terminal in Gitpod:

```bash
cat /proc/sys/kernel/random/uuid
```

papilinou, 2020-11-08:
thanks,done.component tested in CD desktop and fine but no icon.Can't find example how to add.

papilinou, 2020-11-08:
set a pull request for optocoupler.