circuitdiagram/components #102: Please add TITLE BLOCK

papilinou, 2020-11-04:
As in other drawing software a Title Block could be interesting.

papilinou, 2020-11-09:
this is my Title block component design.
Please give me your feeling and remarks before i set a pull request.

[image]
[image]

SamuelFisher, 2020-11-09:
This repository is purely for electronic component symbols currently, so I'm afraid this would not be accepted as a pull request at this time. Please feel free to continue using it in Circuit Diagram Desktop though.

closed 2020-11-09 by SamuelFisher