circuitdiagram/components #110: missing infos in DOCS

papilinou, 2020-11-08:
In DOCS>Components>Render Options>Grid add : in  Gitpod>File>Settings>open preferences>Circuit Diagram>set:
Diagram Show Grid
Render a grid behind the component.(shown in blue).
In DOCS>Components>Render Options>Debug Layout: add in  Gitpod>File>preferences>Circuit Diagram>set:
Render component layout information Diagram Debug Layout.
In DOCS>Components>Finishing:> circuit-diagramcli component my_component.xml -o my_component.cdcom
miss "-" between diagram and cli.> circuit-diagram-cli component my_component.xml -o my_component.cdcom

papilinou, 2020-11-09:
May be not the right place to set this issue,so i set the same on Circuit Diagram.

SamuelFisher, 2020-11-09:
Duplicate of https://github.com/circuitdiagram/circuitdiagram/issues/28.

closed 2020-11-09 by SamuelFisher