circuitdiagram/components #162: Colored Wires

OdinVex, 2021-11-28:
The ability to color wires (and either use solid, dashed, or stripped) would be quite helpful.

khoskinhome, 2021-11-30:
ahhh , you beat me to this comment !

OdinVex, 2021-12-01:
> ahhh , you beat me to this comment ! I was mainly after some extra colours. Brown being really important if trying to indicate 240v AC mains single phase mains wiring .

Delights me to know someone else was hopeful for it too. :)

SamuelFisher, 2023-10-01:
There are more colours coming to the Beta Web Editor which has just been released (including brown).

Please see the below if you would like to try it:

https://circuitdiagram.qa.samuelfisher.co.uk/news/2023/08/web-editor-beta

SamuelFisher, 2023-10-01:
There are more colours coming to the Beta Web Editor which has recently been released (including brown).

Please see the below if you would like to try it:

https://www.circuit-diagram.org/news/2023/08/web-editor-beta