circuitdiagram/components #170: Screen LCD

fschatbot, 2022-02-27:
Here is the YAML Definition I was able to cook up!
```yaml
kind: ComponentConfiguration/v1
metadata:
  guid: a6da97ce-f923-499f-8c62-82251ef3606a
  name: LCD Configuration
  version:
  description:
template:
  name: Integrated Circuit 2
  guid: 2cd49d7c-eb1a-4fe4-9104-c45023e3fc95
properties:
  - name: Header
    value: LCD
  - name: Size
    value: 16
  - name: Pin0
    value: LED-
  - name: Pin1
    value: LED+
  - name: Pin2
    value: DB7
  - name: Pin3
    value: DB6
  - name: Pin4
    value: DB5
  - name: Pin5
    value: DB4
  - name: Pin6
    value: DB3
  - name: Pin7
    value: DB2
  - name: Pin8
    value: DB1
  - name: Pin9
    value: DB0
  - name: Pin10
    value: E
  - name: Pin11
    value: R/W
  - name: Pin12
    value: RS
  - name: Pin13
    value: V0
  - name: Pin14
    value: VDD
  - name: Pin15
    value: VSS
```

fschatbot, 2022-02-27:
I took this configuration from https://docs.arduino.cc/learn/electronics/lcd-displays

S0han, 2022-12-29:
Is there a way to get this component added to the list on the Circuit Diagram Web Editor?

closed 2023-03-21 by SamuelFisher