circuitdiagram/components #178: Variable generator

esluisge, 2022-03-29:
Hi everyone, I am trying to create a new diagram for my physics homework, but it seems like my teacher indicates a variable generator in a particular way. It's a circle with an arrow across it at 45 degrees, it would seem. Can someone tell me how to add this symbol to my diagram? Thank you :) It should look something like this but instead of two lines it should be a circle:

[image]

Corsaka, 2022-03-29:
By generator, do you mean a form of power generator (e.g an AC generator)?

esluisge, 2022-03-30:
Yes