circuitdiagram/components #203: ESP 8266 and Variants

Schnosch, 2022-12-04:


ryanlaycock, 2022-12-30:
Specifically ESP32 would be great 🎉

radius611, 2023-02-14:
Would Love to see esp12-E/F and ESP32

SamuelFisher, 2023-04-01:
ESP32 was added in https://github.com/circuitdiagram/components/pull/228:

https://www.circuit-diagram.org/components/c893311f-5562-483d-9b5f-33647b77cfdb

Gabsang-Lee-Lab, 2023-05-08:
Still would really appreciate ESP8266

metawops, 2023-12-29:
Me too. A "pure" ESP8266 or some ESP8266 dev board, like the Wemos D1 mini would be nice. 👍👍

SamuelFisher, 2024-08-11:
ESP8266 is now available:

https://www.circuit-diagram.org/components/v/extended/8b909330-633e-44b9-8207-8281814cd128/0.3.1

Some further ESP32-XX components can be found here:

https://github.com/circuitdiagram/ext-components/tree/master/kicad/mcu_espressif

closed 2024-08-11 by SamuelFisher