circuitdiagram/components #240: SPDT switch

yeahyeahwhatever234, 2023-07-25:
Please add a single pole dual throw switch

[Symbol](https://soldered.com/productdata/2023/03/schematic-switch-spdt.png)

[Image](https://media.distrelec.com/Web/WebShopImages/portrait_medium/_t/if/taiway-100-SP-1-T1B1M1Q.jpg)

Corsaka, 2023-08-01:
Exists already as a 'Changeover' type switch. Doesn't appear to be functional when 'closed', though.

G1OJS, 2023-08-08:
I've been wanting a DPDT for ages so I have made an XML file for myself (to use with the desktop app) with these below. I will share them as soon as (if I understand correctly) someone approves my first pull request and I can then simply upload them?

[image]