circuitdiagram/components #289: Stepper Motors

fregac, 2024-04-18:
Trying to add steppers to a few schematics I am making, the existing component does not seem to be properly designed.  I don't know why there is an input for supply voltage, and "Input 1-4" doesn't do anything to tell you which coil those go to.  While there are various options I think at the least a bipolar stepper with 4 inputs showing two coils would be useful.

Something like this:
[image]

As opposed to
[image]

biancaquig, 2024-04-29:
I agree

moreirasavio, 2024-10-25:
I agree

SamuelFisher, 2024-10-25:
This is now available:

https://www.circuit-diagram.org/components/8586a4fe-7420-4cae-870d-5f02975e3883

closed 2024-10-25 by SamuelFisher