circuitdiagram/components #305: Filters

taruncousik, 2024-07-30:
Low Pass, Band Pass, Band Stop, High Pass. Useful to have it as a block instead of designing them

MarioSanz, 2024-09-07:
Hi there @taruncousik, I made a component for that, maybe you want to give some feedback on it. This is the band-stop type, but I made it configurable for the other types you mentioned, plus an additional option for switched-capacitor type
[image]

@SamuelFisher Is there a pipeline to submit and review new components?

SamuelFisher, 2024-09-07:
HI @MarioSanz, this looks great.

> Is there a pipeline to submit and review new components?

If you raise a PR in this repo I will prioritise reviewing it. Reviewing is quite time consuming, it will make it a lot easier if you can check things like this in advance before raising a PR:

1. Does it display correctly at different sizes/when resizing
2. Does it display correctly when placed vertically (not only horizontally)
3. Do the connection points (pins where other components and wires can connect to) align correctly, including for different sizes and vertical placement

SamuelFisher, 2024-09-07:
Added in linked PR.

closed 2024-09-07 by SamuelFisher